intel-iot-devkit/tinyb #128: Request implementation of RemoveDevice

daviessm, 2017-10-22:
The recent versions of BlueZ cache all discovered (maybe connected?) devices and return them in a scan even after they've long gone out of range or been switched off. This causes exceptions from tinyb when trying to connect to them, unaware that they aren't actually there.

BlueZ provides the RemoveDevice DBus method and remove_device C method to forget these. Could this be implemented in tinyb?

impala454, 2017-10-23:
BlueZ does not do the caching, dbus does.  Agreed that the feature would be good to add, just wanted to clarify.

petreeftime, 2017-10-23:
This is already implemented, if I understand the request correctly: https://github.com/intel-iot-devkit/tinyb/commit/daa37a0e09bc79c6f21a205b93a211e33042e48b

daviessm, 2017-10-23:
Oh, thanks. I'll give it a go. It isn't in the documentation at http://iotdk.intel.com/docs/master/tinyb/java/classtinyb_1_1BluetoothDevice.html - that's why I didn't see it.

daviessm, 2017-10-23:
Yes, that does what I was looking for.

closed 2017-10-23 by daviessm