intel-iot-devkit/tinyb #129: services missing...

aplastiras, 2017-11-02:
Hi,

I am trying to develop an app showing the heart rate (polar h10) of a person in the screen with Java in Ubuntu. Iam using tinyb library to acces the data from the bluetooth device.

with the gattool i can see all the available services:

irmica@irmica-linux:~$ gatttool -t random --device=F6:49:0D:45:9A:8E --interactive
[F6:49:0D:45:9A:8E][LE]> connect
Attempting to connect to F6:49:0D:45:9A:8E
Connection successful
[F6:49:0D:45:9A:8E][LE]> primary
attr handle: 0x0001, end grp handle: 0x0009 uuid: 00001800-0000-1000-8000-00805f9b34fb
attr handle: 0x000a, end grp handle: 0x000d uuid: 00001801-0000-1000-8000-00805f9b34fb
attr handle: 0x000e, end grp handle: 0x0013 uuid: 0000180d-0000-1000-8000-00805f9b34fb
attr handle: 0x0014, end grp handle: 0x0029 uuid: 0000181c-0000-1000-8000-00805f9b34fb
attr handle: 0x002a, end grp handle: 0x0038 uuid: 0000180a-0000-1000-8000-00805f9b34fb
attr handle: 0x0039, end grp handle: 0x003e uuid: 0000fea5-0000-1000-8000-00805f9b34fb
attr handle: 0x003f, end grp handle: 0x0042 uuid: 0000180f-0000-1000-8000-00805f9b34fb
attr handle: 0x0043, end grp handle: 0x0048 uuid: 6217ff4b-fb31-1140-ad5a-a45545d7ecf3
attr handle: 0x0049, end grp handle: 0xffff uuid: 0000feee-0000-1000-8000-00805f9b34fb
[F6:49:0D:45:9A:8E][LE]> 

but with through my application i can get only 6 of 9.

Device 0 name :Polar H10 1B57E226 address :F6:49:0D:45:9A:8E
Device with name :Polar H10 1B57E226 found. Trying to connect...
connceted :true

serv 0000180a-0000-1000-8000-00805f9b34fb
serv 0000fea5-0000-1000-8000-00805f9b34fb
serv 0000181c-0000-1000-8000-00805f9b34fb
serv 0000180f-0000-1000-8000-00805f9b34fb
serv 0000feee-0000-1000-8000-00805f9b34fb
serv 6217ff4b-fb31-1140-ad5a-a45545d7ecf3

Tinyb is the latest so I suppose it 0.5 version.

My bluez version is 5.37 and I have done everything is mentioned in https://github.com/intel-iot-devkit/tinyb/blob/master/TROUBLESHOOTING.md

Please let me know is something is wrong.
 Thank you in advance

impala454, 2017-11-02:
Can you post your application code?

aplastiras, 2017-11-02:
of course

the code in main that give this result is:

`BluetoothManager manager = BluetoothManager.getBluetoothManager();
	  List<BluetoothDevice> list = manager.getDevices();
	  
	  BluetoothDevice polar = null;
	  
	  for(int i = 0 ; i < list.size() ; i ++) 
		  System.out.println("Device " + i + " name :" + list.get(i).getName() + "address :" + list.get(i).getAddress());
	
	  for(int i = 0 ; i < list.size() ; i ++) {
		  if (list.get(i).getAddress().equals("F6:49:0D:45:9A:8E")) {
		  System.out.println("Device with name :" + list.get(i).getName() + " found. Trying to connect...");
		  polar = (BluetoothDevice) manager.getObject(BluetoothType.DEVICE, list.get(i).getName(), list.get(i).getAddress(), null);
		  System.out.println("connceted :" + polar.connect());
		  System.out.println("paired :" + polar.getPaired());
		  }
	  }
	  

	  List<BluetoothGattService> hrServices =  polar.getServices();
	  
	  
	  for(int i = 0 ; i < hrServices.size() ; i ++) 
		  System.out.println("serv " + hrServices.get(i).getUUID());
           polar.disconnect();
`

impala454, 2017-11-02:
I assume you didn't mean to close the issue?  The problem I see is you call `getServices()` and immediately try to list them.  This is undefined behavior I believe.  You should perform the `getServices()` in a loop to ensure the device has had time to be completely ready.  See the example here:  https://github.com/intel-iot-devkit/tinyb/blob/master/examples/hellotinyb.cpp.

petreeftime, 2017-11-02:
Some of the services are managed by BlueZ and show up as methods or attributes of objects.

impala454, 2017-11-02:
@petreeftime the `0x180d` service is the main heart rate service and should definitely show up as available.  I think honestly @aplastiras gave no time between the `connect()` and `getServices()` call.  There should be a wait in to give the device time, right?

petreeftime, 2017-11-02:
The find should block until the service gets discovered, so that shouldn't be an issue.

impala454, 2017-11-02:
Does the `connect()` call also block?

aplastiras, 2017-11-03:
First of all thank you all for the quick response.

I test the sensor this morning with the HelloTinyB version. This is not my day :( 
Actually it works the same way as before. Here is the results.

```
Address = F6:49:0D:45:9A:8E Name = Polar H10 1B57E226 Connected = false
Found device: Address = F6:49:0D:45:9A:8E Name = Polar H10 1B57E226 Connected = false
Sensor with the provided address connected
Services exposed by device:
UUID: 0000180a-0000-1000-8000-00805f9b34fb
UUID: 0000fea5-0000-1000-8000-00805f9b34fb
UUID: 0000181c-0000-1000-8000-00805f9b34fb
UUID: 0000180f-0000-1000-8000-00805f9b34fb
UUID: 0000feee-0000-1000-8000-00805f9b34fb
UUID: 6217ff4b-fb31-1140-ad5a-a45545d7ecf3
blocked ? : false
This device does not have the heart rate service we are looking for.
```
The desired 0x180d main heart rate service as **impala454** is not in the list. Also impala454 as you can see the device is not blocked. 

I don't know what else I should look for

Please help :)

aplastiras, 2017-11-03:
Solved!!! 

The provlem was the version of blueZ. I had 5.37 and I updated to 5.42. 

Now i have all the data that I need:

```
Address = F6:49:0D:45:9A:8E Name = Polar H10 1B57E226 Connected = false
Found device: Address = F6:49:0D:45:9A:8E Name = Polar H10 1B57E226 Connected = false
Sensor with the provided address connected
Services exposed by device:
Services 1 UUID: 0000180a-0000-1000-8000-00805f9b34fb
	char 0 00002a25-0000-1000-8000-00805f9b34fb
	char 1 00002a24-0000-1000-8000-00805f9b34fb
	char 2 00002a29-0000-1000-8000-00805f9b34fb
	char 3 00002a23-0000-1000-8000-00805f9b34fb
	char 4 00002a28-0000-1000-8000-00805f9b34fb
	char 5 00002a26-0000-1000-8000-00805f9b34fb
	char 6 00002a27-0000-1000-8000-00805f9b34fb
Services 2 UUID: 0000180d-0000-1000-8000-00805f9b34fb
	char 0 00002a38-0000-1000-8000-00805f9b34fb
	char 1 00002a37-0000-1000-8000-00805f9b34fb
		desc 0 00002902-0000-1000-8000-00805f9b34fb
Services 3 UUID: 00001801-0000-1000-8000-00805f9b34fb
	char 0 00002a05-0000-1000-8000-00805f9b34fb
		desc 0 00002902-0000-1000-8000-00805f9b34fb
Services 4 UUID: 0000fea5-0000-1000-8000-00805f9b34fb
	char 0 b5f900a7-aa8d-11e3-9046-0002a5d5c51b
		desc 0 00002902-0000-1000-8000-00805f9b34fb
	char 1 b5f900a8-aa8d-11e3-9046-0002a5d5c51b
Services 5 UUID: 0000181c-0000-1000-8000-00805f9b34fb
	char 0 00002aa2-0000-1000-8000-00805f9b34fb
	char 1 00002a8e-0000-1000-8000-00805f9b34fb
	char 2 00002a98-0000-1000-8000-00805f9b34fb
	char 3 00002a8c-0000-1000-8000-00805f9b34fb
	char 4 00002a80-0000-1000-8000-00805f9b34fb
	char 5 00002a99-0000-1000-8000-00805f9b34fb
	char 6 00002a9f-0000-1000-8000-00805f9b34fb
		desc 0 00002902-0000-1000-8000-00805f9b34fb
	char 7 00002a9a-0000-1000-8000-00805f9b34fb
	char 8 00002a90-0000-1000-8000-00805f9b34fb
	char 9 00002a8a-0000-1000-8000-00805f9b34fb
Services 6 UUID: 0000180f-0000-1000-8000-00805f9b34fb
	char 0 00002a19-0000-1000-8000-00805f9b34fb
		desc 0 00002902-0000-1000-8000-00805f9b34fb
Services 7 UUID: 0000feee-0000-1000-8000-00805f9b34fb
	char 0 fb005c53-02e7-f387-1cad-8acd2d8df0c8
	char 1 fb005c52-02e7-f387-1cad-8acd2d8df0c8
		desc 0 00002902-0000-1000-8000-00805f9b34fb
	char 2 fb005c51-02e7-f387-1cad-8acd2d8df0c8
		desc 0 00002902-0000-1000-8000-00805f9b34fb
Services 8 UUID: 6217ff4b-fb31-1140-ad5a-a45545d7ecf3
	char 0 6217ff4d-91bb-91d0-7e2a-7cd3bda8a1f3
		desc 0 00002902-0000-1000-8000-00805f9b34fb
	char 1 6217ff4c-c8ec-b1fb-1380-3ad986708e2d
Heart Rate service found
```
Thank you all for your support

closed 2017-11-03 by aplastiras