intel-iot-devkit/tinyb #130: Caused by: tinyb.BluetoothException: GDBus.Error:org.bluez.Error.NotSupported: Operation is not supported

tanieee28, 2017-11-06:
Issue raised while invoking writeValue() of BluetoothLeGattCharacterisitic class..

org.eclipse.kura.KuraBluetoothIOException: Error during BLE IO activity.
        at org.eclipse.kura.internal.ble.BluetoothLeGattCharacteristicImpl.writeValue(BluetoothLeGattCharacteristicImpl.java:100)
        at org.eclipse.kura.example.BluetoothLE.BluetoothLeConnection.performScan(BluetoothLeConnection.java:218)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
        at java.util.concurrent.FutureTask.runAndReset(FutureTask.java:308)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.access$301(ScheduledThreadPoolExecutor.java:180)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:294)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617)
        at java.lang.Thread.run(Thread.java:745)
Caused by: tinyb.BluetoothException: GDBus.Error:org.bluez.Error.NotSupported: Operation is not supported
        at tinyb.BluetoothGattCharacteristic.writeValue(Native Method)
        at org.eclipse.kura.internal.ble.BluetoothLeGattCharacteristicImpl.writeValue(BluetoothLeGattCharacteristicImpl.java:98)
        ... 8 more

aplastiras, 2017-11-06:
Hi,

What value are you trying to access? Characteristics have different properties: READ, WRITE, NOTIFY. If the characteristic is READ only or notify then you cannot write on this.

tanieee28, 2017-11-06:
Hi,

I am trying to communicate a raspberry pi 3 bluetooth with the Bluetooth LE 4.0, but while i am trying to write the data to BLE from raspberry bluetooth using Kura.bluetooth.le package classes i am getting the above errror.
By the way, how to check wheather the service is READ only or not.

aplastiras, 2017-11-06:
Hi again,

What kind of sensor (bluetooth device) are you dealing with?
Please check if the service you are looking for is listed [here.](https://www.bluetooth.com/specifications/gatt/services).

In the listed services you can check the property of every characteristic.

tanieee28, 2017-11-06:
Hi again,

I am using the following Services and Characteristics -
public static final UUID SERIAL_SERVICE_UUID = UUID.fromString("0000fff0-0000-1000-8000-00805f9b34fb");
    public static final UUID TX_CHAR_UUID = UUID.fromString("0000fff2-0000-1000-8000-00805f9b34fb");
    public static final UUID RX_CHAR_UUID = UUID.fromString("0000fff1-0000-1000-8000-00805f9b34fb");

I just want to send a text message from my raspberry pi bluetooth to BLE. I have connected them but when i am using writeValue() or readValue(), i am getting the above the error.

tanieee28, 2017-11-07:
Hi

The error is resolved. Actually, it was coming because of the unsupported functionality of the characteristics. The above mentioned characteristic(TX_CHAR_UUID) does not support read and write operations.
As i am communicating with HC-08 BLE, the right characteristic UUID for writing data is **0000ffe1-0000-1000-8000-00805f9b34fb**  which comes under Custom service whose UUID is **0000ffe0-0000-1000-8000-00805f9b34fb**
Thanks for your help @aplastiras .

closed 2017-11-07 by tanieee28

aplastiras, 2017-11-07:
Glad to hear that I helped

Cheers