intel-iot-devkit/tinyb #131: can not get some service

zewenhs, 2017-11-30:
I can not get some services when i use tinyb(hellotinyb.cpp), result as:
[image]

But with gatttool as:
[image]

why there have not 1800 and 1812 (uuid)  service with tinyb? thanks!

zewenhs, 2017-11-30:
I use raspberry pi3 with blueZ 5.47

tonyho, 2017-12-02:
Same issue here. Using the Ubuntu 16.10 with bluez 5.43

aplastiras, 2017-12-02:
Hi to all
Have you checked this guide? 
[https://github.com/intel-iot-devkit/tinyb/blob/master/TROUBLESHOOTING.md](url)

zewenhs, 2017-12-04:
@aplastiras ， hello，

I checked the [TROUBLESHOOTING.md](https://github.com/intel-iot-devkit/tinyb/blob/master/TROUBLESHOOTING.md) and follow it , but not worked for me...
which hardware and software do you used? thank you!

petreeftime, 2017-12-04:
Some services are user by BlueZ and never exposed to others. They are accessible through the methods exposed by BlueZ. In this case 1800 is the the GAP service and it's never exposed. 1812 is the HID service and I think it only gets exposed to root or not at all, since otherwise any program might be capable of reading what is typed in by the user, if the user is using a Bluetooth keyboard.

zewenhs, 2017-12-04:
@petreeftime 
  Thank you for your reply, But why other program cannot get 1812（HID service data）data？  if this is permission issue， there must an method to allow program to discovery this service.
By the way, I can get all of this five services with nrfconnect(app) in android.

vkolotov, 2018-01-11:
The new version of Bluez (v5.48) also removed Battery Service, it is not getting resolved anymore.

https://www.bluetooth.com/specifications/gatt/viewer?attributeXmlFile=org.bluetooth.service.battery_service.xml

Why would they do this... Seems to me odd.

vkolotov, 2018-01-11:
They have introduced a new DBus interface: https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/battery-api.txt

But still... why would they remove a service...