intel-iot-devkit/tinyb #135: SIGSEGV when detaching thread after JVM has exited

pgfisico, 2017-12-23:
A segmentation fault occurs when the JNIEnvContainer destructor detaches the thread and the JVM has already exited.

The cause is `Monitor::lock_without_safepoint_check()` receives a null pointer from `Thread::current()`. `Monitor::ILock()` receives the null pointer as the argument and the segmentation fault occurs when `Monitor::ILock()` dereferences the pointer to access `_MutexEvent`. `_MutexEvent` is at an offset of 0xe4 within `Thread`, which is confirmed by the offending si_addr of 0x000000e4.

Issue also occurs when using Oracle JDK 8u152 Linux ARM 32 Hard Float ABI.

Issue appears similar to #66 and #59.

See also https://bugs.openjdk.java.net/browse/JDK-8033696

**TinyB Version** `0.5.0-28-gac6d308`

### hs_err_pid2261.log
```
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0xb584e300, pid=2261, tid=0xb53b4470
#
# JRE version: OpenJDK Runtime Environment (8.0_151-b12) (build 1.8.0_151-8u151-b12-1~deb9u1-b12)
# Java VM: OpenJDK Client VM (25.151-b12 mixed mode linux-aarch32 )
# Problematic frame:
# V  [libjvm.so+0x2ec300]
#
# Core dump written. Default location: /home/debian/main-board/core or core.2261
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
#

---------------  T H R E A D  ---------------

Current thread is native thread

siginfo: si_signo: 11 (SIGSEGV), si_code: 1 (SEGV_MAPERR), si_addr: 0x000000e4

Registers:
R0=0xb5205218
R1=0x00000000
R2=0x00000001
R3=0x00000001
R4=0xb5205218
R5=0x00000000
R6=0x00183cf6
R7=0xb53b3dd8
R8=0x000c3d18
R9=0xb53b3e5c
R10=0x00000000
R11=0x00000000
R12=0xb59d21dc
R13=0xb53b3dd8
R14=0xb584ee05
R15=0xb584e300

Top of Stack: (sp=0xb53b3dd8)
0xb53b3dd8:   00000001 b5205218 00000000 b6f8fce4
0xb53b3de8:   b53b3df8 000c3d18 b53b3e5c b584ee05
0xb53b3df8:   b59d2000 b6e028c0 b53b3e08 b594f2bb
0xb53b3e08:   b59d2000 b53b4930 b53b3e18 b5783d09
0xb53b3e18:   aa8d9cf8 b53b4930 b53b3e28 aa4363d1
0xb53b3e28:   aa44e000 b59da820 b53b3e38 aa436275
0xb53b3e38:   6a0de79a aa8b80f0 b53b3e48 aa4361cf
0xb53b3e48:   0000000c aa8b80f0 000c3cd6 b6e2b323 

Instructions: (pc=0xb584e300)
0xb584e2e0:   68 3e 18 00 10 20 00 00 8c 23 00 00 46 d8 18 00
0xb584e2f0:   32 d8 18 00 30 06 00 00 2d e9 f8 43 00 af 66 4e
0xb584e300:   d1 f8 e4 50 03 68 7e 44 10 e0 bf f3 5b 8f 50 e8
0xb584e310:   00 2f 9a 42 04 d1 40 e8 00 4e be f1 00 0f f6 d1 

[... 377 lines cut ...]
#10 0xb6e2f322 in __GI___call_tls_dtors () at cxa_thread_atexit_impl.c:155
#11 0xb6f4d5f0 in start_thread (arg=0x0) at pthread_create.c:343
#12 0xb6e9d54a in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:76 from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) up 8
#8  0xaa438278 in JNIEnvContainer::detach (this=0xb52ab2a0) at /home/debian/tinyb_build/tinyb/java/jni/JNIMem.cxx:62
62          vm->DetachCurrentThread();
(gdb) p jni_env
$7 = {env = 0xb5207d30}
(gdb) p env
$8 = (JNIEnv *) 0xb5207d30
(gdb) info thread
  Id   Target Id         Frame
  1    Thread 0xb6e07430 (LWP 4182) "java" __libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:46
* 2    Thread 0xb53b8470 (LWP 4183) "java" Monitor::ILock (this=this@entry=0xb5205200, Self=Self@entry=0x0) at ./src/hotspot/src/share/vm/runtime/mutex.cpp:465
  4    Thread 0xaadff470 (LWP 4185) "java" __libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:46
  5    Thread 0xaadaf470 (LWP 4186) "java" __libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:46
  7    Thread 0xaab74470 (LWP 4188) "java" __libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:46
  8    Thread 0xaaaf4470 (LWP 4189) "java" __libc_do_syscall () at ../sysdeps/unix/sysv/linux/arm/libc-do-syscall.S:46
  10   Thread 0xaa403470 (LWP 4191) "gmain" 0xb6e97580 in poll () at ../sysdeps/unix/syscall-template.S:84
  11   Thread 0xa9c03470 (LWP 4192) "gdbus" 0xb6e97580 in poll () at ../sysdeps/unix/syscall-template.S:84
  12   Thread 0xa9403470 (LWP 4193) "java" 0xaa4381ea in JNIEnvContainer::attach (this=0xb52b1e58) at /home/debian/tinyb_build/tinyb/java/jni/JNIMem.cxx:52

////////// CUT //////////
```

cm-perrs, 2018-10-03:
Is there any progress on this? I have this error on jdk 1.8.0_181-b13.

pgfisico, 2018-11-04:
The segfault here is caused by a bug in HotSpot. It is fixed in JDK 11.

The reason TinyB hit the bug is that it would attach the thread without checking if it was already attached. As a result, for calls that were executing on a thread that started within Java, TinyB would attempt to detach the thread even though it was not a native thread.

For me, this was happening when using notifications. When the C++ code gets a global reference to the callback in Java code, it is re-attaching the thread that originated in Java. This is allowed by JNI and is a no-op. The problem is that it could not detach the thread correctly because the JVM had already cleaned up the thread on shutdown, leading to the null pointer from `Thread::current()`.

https://bugs.openjdk.java.net/browse/JDK-8199012
https://bugs.java.com/view_bug.do?bug_id=8199012