intel-iot-devkit/tinyb #136: Add support for setting advertising data

jockej, 2018-01-16:
I need to be able to set the advertising data for a project I'm working on.
This seems to be possible with Bluez, as per here: https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/advertising-api.txt
but I could not find a method to do it in tinyb (I'm working in Java).

I could give it a go, but I would need some pointers (:-)) on how to do it.

Or maybe somebody else feels like having a try :)

mburger81, 2018-01-24:
Hi @jockej have a look into this issue, #49 

Was you able to implement BLE advertising? We try to do the same thing.

thx

jockej, 2018-01-24:
As far as I understood #49 i solved.

This is not really the same thing, I believe. I want to set the advertising data when advertising. Not read advertisement data from other devices.

mburger81, 2018-01-24:
You wrote "I solved" so you intend you solved your requirement? Can you help?

lachev, 2018-01-24:
Hi there, I'm trying also - as for tinyb - from dbus - c code c++, JNI and up to java. Made my functions and so - but on register _advertisement - it is ok and one instance is used as created LEAdvertiseingManager and LEAdvertisement - but some more stuff must do -  leadvertising_manager1_call_register_advertisement_sync returns ok and get_active_instances become 1 - but I cannot sniff them from my other BLE device. With pyton script in examples is ok (example-advertisement). I'm using BlueZ
 5,48 in experimental

lachev, 2018-01-24:
Note that: for <interface name="org.bluez.LEAdvertisement1"> ; <property name="ManufacturerData" type="a{qv}" access="read"/>  ---- a(qv) not a(sv) for my point of view

lachev, 2018-01-24:
for:
  <interface name="org.bluez.LEAdvertisingManager1">
    <method name="RegisterAdvertisement">
      <arg name="advertisement" type="o" direction="in"/>
      <arg name="options" type="a{sv}" direction="in"/>
    </method>
    <method name="UnregisterAdvertisement">
      <arg name="service" type="o" direction="in"/>
    </method>
    <property name="ActiveInstances" type="y" access="read"/>
    <property name="SupportedInstances" type="y" access="read"/>
    <property name="SupportedIncludes" type="as" access="read"/>
  </interface>
the properties I've got working but on RegisterAdvertisment it says ok but nothing is sending - as far i can see here is some HCI stuf for setting some times and other stuff

mburger81, 2018-01-24:
@lachev WHAT? :) sorry but your message is cryptic for me :)
Where is the LEAdvertisingManager?

jockej, 2018-01-24:
@mburger81 Sorry, I meant "is solved". The last word on that thread seems to be that they got it working.

lachev, 2018-01-24:
when you see /lib/systemd/system/bluetooth.service - inside must be: ExecStart=/usr/local/libexec/bluetooth/bluetoothd --experimental and then in d-feet uoy will see this interface

lachev, 2018-01-24:
<interface name="org.bluez.LEAdvertisingManager1">
    <method name="RegisterAdvertisement">
      <arg name="advertisement" type="o" direction="in"/>
      <arg name="options" type="a{sv}" direction="in"/>
    </method>
    <method name="UnregisterAdvertisement">
      <arg name="service" type="o" direction="in"/>
    </method>
    <property name="ActiveInstances" type="y" access="read"/>
    <property name="SupportedInstances" type="y" access="read"/>
    <property name="SupportedIncludes" type="as" access="read"/>
  </interface>

mburger81, 2018-01-24:
Ok but that's not implemented in tinyb, right?

lachev, 2018-01-24:
[LEAdvertisingManager1.txt](https://github.com/intel-iot-devkit/tinyb/files/1659986/LEAdvertisingManager1.txt)
this is the interface that was unable to be seen

lachev, 2018-01-24:
yes there is no code in tinyb except for auto generated one for LEAdvertisement and nothing for LEAddvertisingManager. Trying to find some solution from 2 weeks

jockej, 2018-01-24:
@lachev Have you tried to implement this? Do you have a fork on github?

lachev, 2018-01-24:
I have a fork – and I feel that I’m going closer – ASAP I have clear code I’ll do it.

From: Joakim Jalap
Sent: 24 януари 2018 г. 15:38
To: intel-iot-devkit/tinyb
Cc: lachev; Mention
Subject: Re: [intel-iot-devkit/tinyb] Add support for setting advertising data(#136)

@lachev Have you tried to implement this? Do you have a fork on github?

mburger81, 2018-01-24:
@lachev beatufil, if you like we can try your fork.

lachev, 2018-01-24:
This days it is full of printouts and commented code

From: mburger81
Sent: 24 януари 2018 г. 15:57
To: intel-iot-devkit/tinyb
Cc: lachev; Mention
Subject: Re: [intel-iot-devkit/tinyb] Add support for setting advertising data(#136)

@lachev beatufil, if you like we can try your fork.

lachev, 2018-01-24:
You can check this https://github.com/carsonmcdonald/bluez-ibeacon (inside bluez-beacon - C written). It is simple and can be added somehow. It seems for me that HCI layer is very important for advertisements. There are some timings and so on. I cannot see it right now over DBUS but can be added like some properties or else.

From: mburger81
Sent: 24 януари 2018 г. 15:57
To: intel-iot-devkit/tinyb
Cc: lachev; Mention
Subject: Re: [intel-iot-devkit/tinyb] Add support for setting advertising data(#136)

@lachev beatufil, if you like we can try your fork.