intel-iot-devkit/tinyb #152: Cmake error library not found for -lglib-2.0

tuzzo77, 2018-10-30:
Hi 
after first command: " cmake .. -DBUILDJAVA=ON" without errors, i lanched "make" but the command generate the following error:

[ 32%] Linking CXX shared library libtinyb.dylib
ld: library not found for -lglib-2.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Can anyone help me?
Thanks

jvervier, 2018-11-30:
Same problem for me, could be interesting to get info :-)

Heckie75, 2019-04-30:
Actually in Ubuntu 19.04 I have installed some additional packages, i.e. 

apt install libglib2.0-dev pkg-config cmake

After this step (and some adjustments for building it in openJDK 10) it works.

tswaehn, 2020-12-29:
`apt install libglib2.0-dev  `
fixes this issue for me.