intel-iot-devkit/tinyb #156: How to handle multiple devices at once

wiesener, 2018-12-27:
Hi I would like to connect to more than one BLE device via tinyb. 
My current workflow is the following:

// First device
manager = BluetoothManager::get_bluetooth_manager();
manager->start_discovery()
auto foundServices = manager->get_services();
this->btdev = manager->find<BluetoothDevice>(...);
btdev->connect()
this->nus_service = btdev->find(&(this->nus_service_uuid), std::chrono::seconds((timeout / 10)));
manager->stop_discovery()
this->nus_char_tx = this->nus_service->find(...);
this->nus_char_rx = this->nus_service->find(...);

// Second device
manager = BluetoothManager::get_bluetooth_manager();
manager->start_discovery()
auto foundServices = manager->get_services();
this->btdev = manager->find<BluetoothDevice>(...);
btdev->connect()
this->nus_service = btdev->find(&(this->nus_service_uuid), std::chrono::seconds((timeout / 10)));
manager->stop_discovery()
this->nus_char_tx = this->nus_service->find(...);
this->nus_char_rx = this->nus_service->find(...);

But as soon as I start discovery again it crashes saying that it is in use. Although I do not store the manager permanently only the device is created and stored in the object.
Is there an example how to handle several connections.

petreeftime, 2018-12-27:
You don't need to stop discovery and start it again unless there is a large gap in time between adding the two devices. You also don't have to get_bluetooth_manager more than once. The underlying object will always be the same.

Unfortunately, this issue is on the BlueZ side, and I don't know if there's a nice way to hide it in TinyB.

It definitely works with multiple devices at once, I tested this at some point, but I didn't add this as an example since it was very specific.

wiesener, 2018-12-27:
Since I've separated it into an Interface class I need to get two times the manager. But I'am only holding the same pointer. 
Would it be wrong to stop discovery and then restart it?

wiesener, 2018-12-27:
"Unfortunately, this issue is on the BlueZ side, and I don't know if there's a nice way to hide it in TinyB."
Which specific part of the issues you mean?

wiesener, 2018-12-27:
Okay cool, leaving the manager in discovering mode makes everything run more performant and I can now connect to several devices at once (So far tested only two of them). But anyway cool. Thanks