intel-iot-devkit/tinyb #157: using tinyb on the peripheral side?

moe5k, 2019-01-22:
We would like to use it on the peripheral side on a raspi 3.
So far i can see tinyb is only for the central side, right?

mikedanielcpt, 2019-03-07:
Hi Moe

I have same question. How can i use tinyb on a helper/peripheral device, from which information is sent to central master device? Were you able to solve this?

Tx
Mike

moe5k, 2019-03-07:
Hi Mike,
i used:
https://github.com/nettlep/gobbledegook
for that.
good luck!
Moe

mikedanielcpt, 2019-03-07:
Hi Moe

Thank you so much.

I take it that there are no official java wrappers available? Have you
developed any java wrappers that you are willing to share?

Kind regards
Micah

moe5k, 2019-03-07:
no , i just used it with c++
Moe

mikedanielcpt, 2019-03-07:
Thanks, would be good to use it as dll from java native interface. Does it
compile to dll?

moe5k, 2019-03-07:
no dll, using a raspi as peripheral

mikedanielcpt, 2019-03-08:
Great. Thanks for the help. Going to try to wrap this in JNI, so that i can
integrate it into my java code.

Warcreed, 2020-08-20:
Hi Mike

Have you found an alternative to JNI or you solved your issue with it? I have the same problem, I need to use rasp as peripheral and use java.

mikedanielcpt, 2020-08-20:
Hi

My project went cold. Gonna pick it up soon again. If you solve the issue ,
can you please share your experience with me? :) I will do the same.

Kind regards
Mike