intel-iot-devkit/tinyb #161: Tinyb does not pick up all devices

mikedanielcpt, 2019-02-27:
Hi All

Thank you soooo much for taking time to read and reply on my problem. 

I have read another post on github that is remotely similar but does not address my issue. I have also been reading many threads on different sites whilst trying to uncover the cause of my problem. 

I am using the tinyb java library from Intel get two pi's (3b+ and a zero w) to communicate via BLE.

Using "sudo hcitool lescan" from each pi, I am able to detect the other pi (and 10 other devices in the apartment block). However, when using below code from 3b+ (360 loops with 10sec sleeps in between), i find up to 23 other apartment block devices but not the pi zero w. When comparing the two device lists I noticed that only 1 device (from somewhere in apartment block) is common to both listings. So it seems that tinyb is picking up a ble device, just not my pi zero w.

Furthermore, I am running bluez v5.43. i tried to upgrade to v5.48 and v5.50. Everything seems to work fine during the make process until i run "make -j4". Whilst executing the command, the pi3b+ reboots. See [u]error 1[/u] below. From then the install does not want to continue. See [u]error 2[/u] below. 

//----------------error 1 tag

pi@micah3bplus:~/Documents/Apps/OS/bluez-5.50 $ make -j4
  GEN      lib/bluetooth/hci.h
  GEN      lib/bluetooth/hci_lib.h
  GEN      lib/bluetooth/bluetooth.h
  GEN      lib/bluetooth/sco.h
  GEN      lib/bluetooth/sdp.h
  GEN      lib/bluetooth/l2cap.h
  GEN      lib/bluetooth/sdp_lib.h
  GEN      lib/bluetooth/rfcomm.h
  GEN      lib/bluetooth/cmtp.h
  GEN      src/builtin.h
  GEN      lib/bluetooth/bnep.h
  GEN      lib/bluetooth/hidp.h
  GEN      obexd/src/builtin.h
make --no-print-directory all-am
  GEN      src/bluetooth.service

(reboots or terminates ssh session, depending if ssh used or direct connection to pi via monitor and keyboard)

//-----------error2 tag
pi@mike3bplus:~/Documents/Apps/OS/bluez-5.50 $ sudo make install
make: *** No rule to make target 'install'.  Stop.
pi@mike3bplus:~/Documents/Apps/OS/bluez-5.50 $ ^C
pi@mike3bplus:~/Documents/Apps/OS/bluez-5.50 $ make -j4
make: *** No targets.  Stop.

I am not sure if upgrading from bluez 5.43 will help with detecting my pi zero from 3b+ using tinyb. This is the primary problem i need to solve. Getting to the root of the upgrading problem will be a bonus though if it does not impact the tinyb library function, else it is a necessity.

Tx
Mike

//=================CODE======================//

static BluetoothDevice getDevice(String address, BluetoothManager manager) throws InterruptedException
    {
        BluetoothDevice sensor = null;
        
        for (int i = 0; (i < 360) && running; ++i)
        {
            List<BluetoothDevice> list = manager.getDevices();
          
            for (BluetoothDevice device : list)
            {
                if (device.getAddress().equals(address))
                {
                    sensor = device;
                    System.out.println("MATCH : Target - " + address +" - Found - " + device.getAddress() + "!!!!!!!!");
                }
                else
                {
                    System.out.println("Not match : Target - " + address +" - Found - " + device.getAddress() );
                }
            }

            if (sensor != null)  
            {
                return sensor;
            }
            Thread.sleep(10000);
        }
        return null;
    }

mikedanielcpt, 2019-02-27:
So made some progress....

I managed to install bluez v5.50....I  needed a better power supply which i got. Same issue though, tinyb on pi3+ is not picking up my zero w, even though lescan is .

mikedanielcpt, 2019-03-07:
Hi

I have resolved this issue.

Tx
Mike

spriggsy, 2019-09-01:
Not come up against this error yet but in case i do, and for any others, can you explain what you done to fix the issue?

thanks