intel-iot-devkit/tinyb #163: installation: make error Ubuntu 18.04

castroonoh, 2019-04-30:
I'm trying to install tinyb on my PC (ubuntu 18.04 os) but I receive this error during the make command:

Generating JNI headers..
Error: Could not find class file for 'tinyb.BluetoothAdapter'.
java/CMakeFiles/tinybjar.dir/build.make:93: recipe for target 'tinybjar' failed
make[2]: *** [tinybjar] Error 1
CMakeFiles/Makefile2:87: recipe for target 'java/CMakeFiles/tinybjar.dir/all' failed
make[1]: *** [java/CMakeFiles/tinybjar.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

I've searched in the folders and the .class files are in that position so I cannot understand why I have this issue. Could you help me?

bwssytems, 2019-05-06:
I second this as I have this error as well. I made sure to build the cpp side first and install that then turn the java binding on and compile and get the same as above on Linux Mint 19.1:

#root@/git/tinyb/build# make
Scanning dependencies of target tinybjar
[  2%] Building Java objects for tinybjar.jar
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
[  4%] Generating CMakeFiles/tinybjar.dir/java_class_filelist
[  6%] Creating Java archive tinyb.jar
Generating JNI headers..
Error: Could not find class file for 'tinyb.BluetoothAdapter'.
java/CMakeFiles/tinybjar.dir/build.make:93: recipe for target 'tinybjar' failed
make[2]: *** [tinybjar] Error 1
CMakeFiles/Makefile2:87: recipe for target 'java/CMakeFiles/tinybjar.dir/all' failed
make[1]: *** [java/CMakeFiles/tinybjar.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

bwssytems, 2019-05-07:
So went through a lot of tries on building and determined that javah is not working in 1.8 or above. Switched to Java 11 and changed the cmake to use javac -h instead. Had to have the CMakeLists.txt to make it work though. but now have the library. Here is the updated CMakeLists.txt that needs to be updated in the java directory before you run 'sudo cmake -DBUILDJAVA=ON ..'

'''
find_package(Java REQUIRED)
include(UseJava)

if (NOT DEFINED $ENV{JAVA_HOME_NATIVE})
  set (JAVA_HOME_NATIVE $ENV{JAVA_HOME})
  set (JAVAC $ENV{JAVA_HOME}/bin/javac)
  set (JAR $ENV{JAVA_HOME}/bin/jar)
else ()
  set (JAVAC $ENV{JAVA_HOME_NATIVE}/bin/javac)
  set (JAR $ENV{JAVA_HOME_NATIVE}/bin/jar)
endif ()

# Check that we can use javah - make sure to delete the link to javah
if(NOT Java_JAVAH_EXECUTABLE)
	set (Java_JAVAH_EXECUTABLE  /usr/lib/jvm/default-java/bin/javac)
endif(NOT Java_JAVAH_EXECUTABLE)

set(CMAKE_JNI_TARGET TRUE)
file(GLOB JAVA_SOURCES "*.java")
add_jar(tinybjar ${JAVA_SOURCES}
                  MANIFEST ${CMAKE_CURRENT_BINARY_DIR}/manifest.txt
                  OUTPUT_NAME tinyb
)

set(JAVA_CLASSES /home/bsamuels/Downloads/tinyb-0.5.1/java/BluetoothAdapter.java
                 ~/tinyb-0.5.1/java/BluetoothCallback.java
                 ~/tinyb-0.5.1/java/BluetoothDevice.java
                 ~/tinyb-0.5.1/java/BluetoothEvent.java
                 ~/tinyb-0.5.1/java/BluetoothGattCharacteristic.java
                 ~/tinyb-0.5.1/java/BluetoothGattService.java
                 ~/tinyb-0.5.1/java/BluetoothGattDescriptor.java
                 ~/tinyb-0.5.1/java/BluetoothManager.java
                 ~/tinyb-0.5.1/java/BluetoothObject.java
                 ~/tinyb-0.5.1/java/BluetoothType.java
                 ~/tinyb-0.5.1/java/BluetoothException.java
                 ~/tinyb-0.5.1/java/ObjectArgCallback.java
                 ~/tinyb-0.5.1/java/ObjectArrayArgCallback.java)
add_custom_command (TARGET tinybjar
  POST_BUILD
  COMMAND ${CMAKE_COMMAND} -E echo "Generating JNI headers.."
  WORKING_DIRECTORY "${CMAKE_CURRENT_BINARY_DIR}/${CMAKE_FILES_DIRECTORY}/tinybjar.dir"
  COMMAND pwd
  COMMAND ${Java_JAVAH_EXECUTABLE} -h jni/ ${JAVA_CLASSES}
)

set(JNI_HEADER_PATH "${CMAKE_CURRENT_BINARY_DIR}/${CMAKE_FILES_DIRECTORY}/tinybjar.dir/jni")
install (FILES ${CMAKE_CURRENT_BINARY_DIR}/tinyb.jar DESTINATION ${CMAKE_INSTALL_LIBDIR}/../lib/java)

add_subdirectory (jni)
'''

mpbarnwell, 2019-06-25:
I was able to overcome this issue by explicitly setting the JAVA_HOME environmental variable then re-running cmake:

```
export JAVA_HOME=/usr/lib/jvm/java-8-openjdk-amd64/
mkdir build
cd build
cmake .. -DBUILDJAVA=ON -DCMAKE_INSTALL_PREFIX=/usr
make
```

The output for those interested:

```
Scanning dependencies of target tinybjar
[  2%] Building Java objects for tinybjar.jar
Note: Some input files use unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
[  4%] Generating CMakeFiles/tinybjar.dir/java_class_filelist
[  6%] Creating Java archive tinyb.jar
Generating JNI headers..
[  6%] Built target tinybjar
Scanning dependencies of target tinyb
[  8%] Building CXX object src/CMakeFiles/tinyb.dir/BluetoothObject.cpp.o
[ 10%] Building CXX object src/CMakeFiles/tinyb.dir/BluetoothEvent.cpp.o
[ 12%] Building CXX object src/CMakeFiles/tinyb.dir/BluetoothManager.cpp.o
[ 14%] Building CXX object src/CMakeFiles/tinyb.dir/BluetoothAdapter.cpp.o
[ 16%] Building CXX object src/CMakeFiles/tinyb.dir/BluetoothDevice.cpp.o
[ 18%] Building CXX object src/CMakeFiles/tinyb.dir/BluetoothGattService.cpp.o
[ 20%] Building CXX object src/CMakeFiles/tinyb.dir/BluetoothGattCharacteristic.cpp.o
[ 22%] Building CXX object src/CMakeFiles/tinyb.dir/BluetoothGattDescriptor.cpp.o
[ 24%] Building CXX object src/CMakeFiles/tinyb.dir/BluetoothUUID.cpp.o
/home/michael/src/tinyb/src/BluetoothUUID.cpp: In member function ‘std::__cxx11::string tinyb::BluetoothUUID::get_string()’:
/home/michael/src/tinyb/src/BluetoothUUID.cpp:59:39: warning: format ‘%lx’ expects argument of type ‘long unsigned int’, but argument 5 has type ‘long long unsigned int’ [-Wformat=]
/home/michael/src/tinyb/src/BluetoothUUID.cpp:56:9:
         ((uuid[0] >> 16) & 0xFFFFULL),
         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~  
/home/michael/src/tinyb/src/BluetoothUUID.cpp:59:39:
         (uuid[1] & ~(0xFFFFULL << 48)));
                                       ^
/home/michael/src/tinyb/src/BluetoothUUID.cpp:59:39: warning: format ‘%lx’ expects argument of type ‘long unsigned int’, but argument 6 has type ‘long long unsigned int’ [-Wformat=]
/home/michael/src/tinyb/src/BluetoothUUID.cpp:59:39: warning: format ‘%lx’ expects argument of type ‘long unsigned int’, but argument 8 has type ‘long long unsigned int’ [-Wformat=]
/home/michael/src/tinyb/src/BluetoothUUID.cpp: In member function ‘std::__cxx11::string tinyb::BluetoothUUID::get_short_string()’:
/home/michael/src/tinyb/src/BluetoothUUID.cpp:69:42: warning: format ‘%lx’ expects argument of type ‘long unsigned int’, but argument 4 has type ‘uint32_t {aka unsigned int}’ [-Wformat=]
             snprintf(u, 9, "%04lx", suuid);
                                          ^
/home/michael/src/tinyb/src/BluetoothUUID.cpp:71:42: warning: format ‘%lx’ expects argument of type ‘long unsigned int’, but argument 4 has type ‘uint32_t {aka unsigned int}’ [-Wformat=]
             snprintf(u, 9, "%08lx", suuid);
                                          ^
[ 26%] Building CXX object src/CMakeFiles/tinyb.dir/tinyb_utils.cpp.o
[ 28%] Building C object src/CMakeFiles/tinyb.dir/generated-code.c.o
[ 30%] Building C object src/CMakeFiles/tinyb.dir/version.c.o
[ 32%] Linking CXX shared library libtinyb.so
[ 32%] Built target tinyb
Scanning dependencies of target javatinyb
[ 34%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/BluetoothAdapter.cxx.o
[ 36%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/BluetoothDevice.cxx.o
[ 38%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/BluetoothEvent.cxx.o
[ 40%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/BluetoothGattCharacteristic.cxx.o
[ 42%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/BluetoothGattDescriptor.cxx.o
[ 44%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/BluetoothGattService.cxx.o
[ 46%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/BluetoothManager.cxx.o
[ 48%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/BluetoothObject.cxx.o
[ 50%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/JNIMem.cxx.o
/home/michael/src/tinyb/java/jni/JNIMem.cxx: In function ‘jint JNI_OnLoad(JavaVM*, void*)’:
/home/michael/src/tinyb/java/jni/JNIMem.cxx:30:39: warning: unused parameter ‘reserved’ [-Wunused-parameter]
 jint JNI_OnLoad(JavaVM *initVM, void *reserved) {
                                       ^~~~~~~~
[ 52%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/helper.cxx.o
[ 54%] Linking CXX shared library libjavatinyb.so
[ 54%] Built target javatinyb
[... 20 lines cut ...]
[ 78%] Building CXX object examples/CMakeFiles/hellotinyb.dir/hellotinyb.cpp.o
[ 80%] Linking CXX executable hellotinyb
[ 80%] Built target hellotinyb
Scanning dependencies of target esstinyb
[ 82%] Building CXX object examples/CMakeFiles/esstinyb.dir/esstinyb.cpp.o
[ 84%] Linking CXX executable esstinyb
[ 84%] Built target esstinyb
Scanning dependencies of target uuid
[ 86%] Building CXX object examples/CMakeFiles/uuid.dir/uuid.cpp.o
[ 88%] Linking CXX executable uuid
[ 88%] Built target uuid
Scanning dependencies of target list_mfg
[ 90%] Building CXX object examples/CMakeFiles/list_mfg.dir/list_mfg.cpp.o
[ 92%] Linking CXX executable list_mfg
[ 92%] Built target list_mfg
Scanning dependencies of target asynctinyb
[ 94%] Building CXX object examples/CMakeFiles/asynctinyb.dir/asynctinyb.cpp.o
[ 96%] Linking CXX executable asynctinyb
[ 96%] Built target asynctinyb
Scanning dependencies of target notifications
[ 98%] Building CXX object examples/CMakeFiles/notifications.dir/notifications.cpp.o
[100%] Linking CXX executable notifications
[100%] Built target notifications

```

gpietro, 2020-06-16:
@bwssytems I did try your solution without success... (I'm running Java 11 on raspberry pi 3)

```
[ 34%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/BluetoothAdapter.cxx.o
/home/pi/dev/libs/tinyb/java/jni/BluetoothAdapter.cxx:29:10: fatal error: tinyb_BluetoothAdapter.h: No such file or directory
 #include "tinyb_BluetoothAdapter.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [java/jni/CMakeFiles/javatinyb.dir/build.make:63: java/jni/CMakeFiles/javatinyb.dir/BluetoothAdapter.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:148: java/jni/CMakeFiles/javatinyb.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

@mpbarnwell also tired setting Java 8 with `/usr/lib/jvm/java-8-openjdk-armhf/jre/bin/java` but it couldn't find the class.

😭

davidzof, 2020-09-18:
@gpietro Worked ok for me with a Java 8 JDK. No go with Java 11.

export JAVA_HOME=/usr/lib/jvm/java-8-openjdk-amd64/
cd build
cmake .. -DBUILDJAVA=ON -DCMAKE_INSTALL_PREFIX=/usr
make
sudo make install

on Ubuntu 18.4

a333klm, 2020-11-20:
#164 might solve this.

franciscomirasg, 2021-04-30:
Hi,
I follow the instruction above but I am getting this error
`
spectre@skyranger2:~/tinyb-0.5.1/build$ sudo make
Generating JNI headers..
/home/spectre/tinyb-0.5.1/build/java/CMakeFiles/tinybjar.dir
Usage: 
  javah [options] <classes>
where [options] include:
  -o <file>                Output file (only one of -d or -o may be used)
  -d <dir>                 Output directory
  -v  -verbose             Enable verbose output
  -h  --help  -?           Print this message
  -version                 Print version information
  -jni                     Generate JNI-style header file (default)
  -force                   Always write output files
  -classpath <path>        Path from which to load classes
  -cp <path>               Path from which to load classes
  -bootclasspath <path>    Path from which to load bootstrap classes
<classes> are specified with their fully qualified names
(for example, java.lang.Object).
[  6%] Built target tinybjar
[ 32%] Built target tinyb
[ 34%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/BluetoothAdapter.cxx.o
/home/spectre/tinyb-0.5.1/java/jni/BluetoothAdapter.cxx:29:10: fatal error: tinyb_BluetoothAdapter.h: No existe el archivo o el directorio
   29 | #include "tinyb_BluetoothAdapter.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [java/jni/CMakeFiles/javatinyb.dir/build.make:63: java/jni/CMakeFiles/javatinyb.dir/BluetoothAdapter.cxx.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:248: java/jni/CMakeFiles/javatinyb.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
`
On Ubuntu 20.04

AntistesSacrorum, 2021-05-04:
Make sure you have jdk and not just jre. BTW you don't need to install it, just download an archive from here, for example: https://adoptopenjdk.net/releases.html and extract to some folder.
One more point, make sure you have cmake v3.11 and above versions below doesn't support java 10+ (javah was replaced by javac -h")

dejanovicnnd-symphony, 2022-01-27:
Hello there!

I still cannot understand how am I missing this file `tiny_BluetoothAdapt.h` so `make` is not succeeding.. Also, where is this file coming from?

I get this:
```
[ 34%] Building CXX object java/jni/CMakeFiles/javatinyb.dir/BluetoothAdapter.cxx.o
/Users/nnd/Projects/MyProjects/tinyb/java/jni/BluetoothAdapter.cxx:29:10: fatal error: 'tinyb_BluetoothAdapter.h' file not found
#include "tinyb_BluetoothAdapter.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [java/jni/CMakeFiles/javatinyb.dir/BluetoothAdapter.cxx.o] Error 1
make[1]: *** [java/jni/CMakeFiles/javatinyb.dir/all] Error 2
make: *** [all] Error 2
```

bwssytems, 2022-03-31:
So updating what works for me now that I had to go back and use tinyb again:

OpenJdk 1.11
Linux Mint 20.3 Una
cmake version 3.16.3
GNU Make 4.2.1

running from the clone from git and following the build instructions for Java

Change the file java/CMakeLists.txt to this and replace <PathTo> to the actual path
```
find_package(Java REQUIRED)
include(UseJava)

if (NOT DEFINED $ENV{JAVA_HOME_NATIVE})
  set (JAVA_HOME_NATIVE $ENV{JAVA_HOME})
  set (JAVAC $ENV{JAVA_HOME}/bin/javac)
  set (JAR $ENV{JAVA_HOME}/bin/jar)
else ()
  set (JAVAC $ENV{JAVA_HOME_NATIVE}/bin/javac)
  set (JAR $ENV{JAVA_HOME_NATIVE}/bin/jar)
endif ()

set(CMAKE_JNI_TARGET TRUE)
file(GLOB JAVA_SOURCES "*.java")
add_jar(tinybjar ${JAVA_SOURCES}
                  MANIFEST ${CMAKE_CURRENT_BINARY_DIR}/manifest.txt
                  OUTPUT_NAME tinyb
)

set(JAVA_CLASSES /home/bsamuels/git/tinyb/java/BluetoothAdapter.java
                <PathTo>/tinyb/java/BluetoothCallback.java
                 <PathTo>/tinyb/java/BluetoothDevice.java
                 <PathTo>/tinyb/java/BluetoothEvent.java
                 <PathTo>/git/tinyb/java/BluetoothGattCharacteristic.java
                 <PathTo>/tinyb/java/BluetoothGattService.java
                 <PathTo>/tinyb/java/BluetoothGattDescriptor.java
                 <PathTo>/tinyb/java/BluetoothManager.java
                 <PathTo>/tinyb/java/BluetoothObject.java
                <PathTo>/tinyb/java/BluetoothType.java
                 <PathTo>/tinyb/java/BluetoothException.java
                 <PathTo>/tinyb/java/ObjectArgCallback.java
                 <PathTo>/tinyb/java/ObjectArrayArgCallback.java)
add_custom_command (TARGET tinybjar
  POST_BUILD
  COMMAND ${CMAKE_COMMAND} -E echo "Generating JNI headers.."
  WORKING_DIRECTORY "${CMAKE_CURRENT_BINARY_DIR}/${CMAKE_FILES_DIRECTORY}/tinybjar.dir"
  COMMAND ${JAVAC} -h jni/ ${JAVA_CLASSES}
)

set(JNI_HEADER_PATH "${CMAKE_CURRENT_BINARY_DIR}/${CMAKE_FILES_DIRECTORY}/tinybjar.dir/jni")
install (FILES ${CMAKE_CURRENT_BINARY_DIR}/tinyb.jar DESTINATION ${CMAKE_INSTALL_LIBDIR}/../lib/java)

add_subdirectory (jni)
```

Need to Change Java file java/BluetoothManager.java and comment out API version code like so:
```

    public static synchronized BluetoothManager getBluetoothManager() throws RuntimeException, BluetoothException
    {
        if (inst == null)
        {
/**
            String nativeAPIVersion = getNativeAPIVersion();
            String APIVersion = BluetoothManager.class.getPackage().getSpecificationVersion();
            if (APIVersion.equals(nativeAPIVersion) == false) {
                String[] nativeAPIVersionCode = nativeAPIVersion.split("\\D");
                String[] APIVersionCode = APIVersion.split("\\D");
                if (APIVersionCode[0].equals(nativeAPIVersionCode[0]) == false) {
                    if (Integer.valueOf(APIVersionCode[0]) < Integer.valueOf(nativeAPIVersionCode[0]))
                        throw new RuntimeException("Java library is out of date. Please update the Java library.");
                    else throw new RuntimeException("Native library is out of date. Please update the native library.");
                }
                else if (APIVersionCode[0].equals("0") == true) {
                    if (Integer.valueOf(APIVersionCode[1]) < Integer.valueOf(nativeAPIVersionCode[1]))
                        throw new RuntimeException("Java library is out of date. Please update the Java library.");
                    else throw new RuntimeException("Native library is out of date. Please update the native library.");
                }
                else if (Integer.valueOf(APIVersionCode[1]) < Integer.valueOf(nativeAPIVersionCode[1]))
                    System.err.println("Java library is out of date. Please update the Java library.");
                else System.err.println("Native library is out of date. Please update the native library.");
            }
*/
            inst = new BluetoothManager();
            inst.init();
        }
        return inst;
    }

    protected void finalize()
    {
        delete();
    }
```

victorex27, 2022-04-04:
> So updating what works for me now that I had to go back and use tinyb again:
> 
> OpenJdk 1.11 Linux Mint 20.3 Una cmake version 3.16.3 GNU Make 4.2.1
> 
> running from the clone from git and following the build instructions for Java
> 
> Change the file java/CMakeLists.txt to this and replace to the actual path
> 
> ```
> find_package(Java REQUIRED)
> include(UseJava)
> 
> if (NOT DEFINED $ENV{JAVA_HOME_NATIVE})
>   set (JAVA_HOME_NATIVE $ENV{JAVA_HOME})
>   set (JAVAC $ENV{JAVA_HOME}/bin/javac)
>   set (JAR $ENV{JAVA_HOME}/bin/jar)
> else ()
>   set (JAVAC $ENV{JAVA_HOME_NATIVE}/bin/javac)
>   set (JAR $ENV{JAVA_HOME_NATIVE}/bin/jar)
> endif ()
> 
> 
> set(CMAKE_JNI_TARGET TRUE)
> file(GLOB JAVA_SOURCES "*.java")
> add_jar(tinybjar ${JAVA_SOURCES}
>                   MANIFEST ${CMAKE_CURRENT_BINARY_DIR}/manifest.txt
>                   OUTPUT_NAME tinyb
> )
> 
> set(JAVA_CLASSES <PathTo>/tinyb/java/BluetoothAdapter.java
>                 <PathTo>/tinyb/java/BluetoothCallback.java
>                  <PathTo>/tinyb/java/BluetoothDevice.java
>                  <PathTo>/tinyb/java/BluetoothEvent.java
>                  <PathTo>/git/tinyb/java/BluetoothGattCharacteristic.java
>                  <PathTo>/tinyb/java/BluetoothGattService.java
>                  <PathTo>/tinyb/java/BluetoothGattDescriptor.java
>                  <PathTo>/tinyb/java/BluetoothManager.java
>                  <PathTo>/tinyb/java/BluetoothObject.java
>                 <PathTo>/tinyb/java/BluetoothType.java
>                  <PathTo>/tinyb/java/BluetoothException.java
>                  <PathTo>/tinyb/java/ObjectArgCallback.java
>                  <PathTo>/tinyb/java/ObjectArrayArgCallback.java)
> add_custom_command (TARGET tinybjar
>   POST_BUILD
>   COMMAND ${CMAKE_COMMAND} -E echo "Generating JNI headers.."
>   WORKING_DIRECTORY "${CMAKE_CURRENT_BINARY_DIR}/${CMAKE_FILES_DIRECTORY}/tinybjar.dir"
>   COMMAND ${JAVAC} -h jni/ ${JAVA_CLASSES}
> )
> 
> set(JNI_HEADER_PATH "${CMAKE_CURRENT_BINARY_DIR}/${CMAKE_FILES_DIRECTORY}/tinybjar.dir/jni")
> install (FILES ${CMAKE_CURRENT_BINARY_DIR}/tinyb.jar DESTINATION ${CMAKE_INSTALL_LIBDIR}/../lib/java)
> 
> add_subdirectory (jni)
> ```
> 
> Need to Change Java file java/BluetoothManager.java and comment out API version code like so:
> 
> ```
> 
>     public static synchronized BluetoothManager getBluetoothManager() throws RuntimeException, BluetoothException
>     {
>         if (inst == null)
>         {
> /**
>             String nativeAPIVersion = getNativeAPIVersion();
>             String APIVersion = BluetoothManager.class.getPackage().getSpecificationVersion();
>             if (APIVersion.equals(nativeAPIVersion) == false) {
>                 String[] nativeAPIVersionCode = nativeAPIVersion.split("\\D");
>                 String[] APIVersionCode = APIVersion.split("\\D");
>                 if (APIVersionCode[0].equals(nativeAPIVersionCode[0]) == false) {
>                     if (Integer.valueOf(APIVersionCode[0]) < Integer.valueOf(nativeAPIVersionCode[0]))
>                         throw new RuntimeException("Java library is out of date. Please update the Java library.");
>                     else throw new RuntimeException("Native library is out of date. Please update the native library.");
>                 }
>                 else if (APIVersionCode[0].equals("0") == true) {
>                     if (Integer.valueOf(APIVersionCode[1]) < Integer.valueOf(nativeAPIVersionCode[1]))
>                         throw new RuntimeException("Java library is out of date. Please update the Java library.");
>                     else throw new RuntimeException("Native library is out of date. Please update the native library.");
>                 }
>                 else if (Integer.valueOf(APIVersionCode[1]) < Integer.valueOf(nativeAPIVersionCode[1]))
>                     System.err.println("Java library is out of date. Please update the Java library.");
>                 else System.err.println("Native library is out of date. Please update the native library.");
>             }
> */
>             inst = new BluetoothManager();
>             inst.init();
>         }
>         return inst;
>     }
> 
>     protected void finalize()
>     {
>         delete();
>     }
> ```

Worked for me. After so many days of stress. Thanks.