intel-iot-devkit/tinyb #168: Project unmaintained?

IOOOTAlan, 2019-08-06:
Greetings,

it seems like it has been some time since this project has seen some activity from its maintainers (commits, issue discussions, reviews on pull requests, etc.). Is it still maintained?

petero-dk, 2019-11-09:
I would seem that the development is still going, but commits are on the kernel.org git repository:

https://git.kernel.org/pub/scm/bluetooth/bluez.git

(I does however look like there was a significant pause for the last year and a half)

IOOOTAlan, 2019-11-11:
@petero-dk that is the upstream bluez git repository and it doesn't seem like tinyb got merged inside of it, so the question still remains.

petreeftime, 2019-11-13:
I don't think Intel is actively maintaining this project anymore. Since I no longer work for Intel I don't have the necessary gear and time to continue developing and testing this.

IOOOTAlan, 2019-11-14:
Thanks @petreeftime, I guess that explains the lack of recent activity.

I'll keep this issue open, if someone from Intel wants to add an official statement regarding the project status/direction.