intel-iot-devkit/tinyb #176: GDBus Error on new discoveries

giuseppemr, 2020-10-09:
### Good morning everyone,
I am actually using **tinyb library for Java** on a **Raspberry Pi 3+** to scan for BLE devices. 
After a period of time (it can be minutes or hours), new scans make **the system blocks in a non recoverable loop with this error**:

```
2020-10-09T08:27:16,310 [Thread-896] ERROR o.e.k.i.b.BluetoothLeAdapterImpl - Stop discovery failed
tinyb.BluetoothException: GDBus.Error:org.freedesktop.DBus.Error.ServiceUnknown: The name :1.7 was not provided by any .service files
        at tinyb.BluetoothAdapter.stopDiscovery(Native Method)
        at org.eclipse.kura.internal.ble.BluetoothLeAdapterImpl$BluetoothFuture.run(BluetoothLeAdapterImpl.java:266)
        at java.lang.Thread.run(Thread.java:748)
```

I tried to stop and restart my program, restart the bluetooth service, but the only solution is to completely reboot the system.
Any help is really appreciated. Thanks.

tommyNy, 2020-11-23:
@giuseppemr check solution with power on/off adapter

giuseppemr, 2020-11-23:
@tommyNy thank you for replying. Could you please send me the link of this solution?

tommyNy, 2020-11-23:
@giuseppemr check bluetoothctl CLI tool for Linux