intel-iot-devkit/tinyb #183: Error when adding tinyB on Maven

ridhakchouk, 2021-06-15:
Hello I am having problems adding tinyB to Maven, according to one of the comments on this post [Stackoverflow post](https://stackoverflow.com/questions/67431634/how-to-add-and-artifact-located-at-openhab-repository-with-maven) it seems that the artifacts have been hosted on bintray which has gone down.

Is it possible to have a fix ?

IOOOTAlan, 2021-06-16:
The project is unmaintained and is not updated since 4 years, so there's probably no one that will re-upload this project artifacts anywhere.
However, you can probably find repackaged tinyb jars from other projects, e.g. on OpenHAB libraries m2 repository: https://openhab.jfrog.io/openhab/libs-release/, under `org.openhab.osgiify/intel-iot-devkit.tinyb`.
I would not depend on any of them for anything serious though, as there's a high chance that the projects making use of these repackaged jars are already migrating away from tinyb (OpenHAB is going directly on bluez, for example).

ridhakchouk, 2021-06-16:
Thank you for your answer. 

So I'm using this dependency but I still have an error : Cannot resolve org.openhab.osgiify:intel-iot-devkit.tinyb:0.5.1
I start to have problems using this library with maven.

I am trying to develop an application that reads information from a sensor connected to a BLE device.

`
        <dependency>
            <groupId>org.openhab.osgiify</groupId>
            <artifactId>intel-iot-devkit.tinyb</artifactId>
            <version>0.5.1</version>
        </dependency>`

Do you know how to solve this ?

IOOOTAlan, 2021-06-16:
Basically, you must add a `repository` section in your POM or in a parent pointing at the OpenHAB repository URL i linked previously. Then, the dependency should be resolved.
See https://maven.apache.org/pom.html#Repository

ridhakchouk, 2021-06-16:
I'm trying to add the repository section in my POM. Can you help me to do this correctly ?

This is what I add in my POM file for the repository section :
    

    <distributionManagement>
        <repository>
            <id>corp1</id>
            <name>openHabTinyB</name>
            <url>https://openhab.jfrog.io/openhab/libs-release/</url>
        </repository>
    </distributionManagement>
    
    This is the dependency : 
    
    
        <dependency>
            <groupId>org.openhab.osgiify</groupId>
            <artifactId>intel-iot-devkit.tinyb</artifactId>
            <version>0.5.1</version>
        </dependency>

IOOOTAlan, 2021-06-16:
You've put the repository in the wrong section: instead of `distributionManagement` you should use `repositories`.

ridhakchouk, 2021-06-16:
Thank you this is working !

closed 2021-06-16 by ridhakchouk